hdl/containers #17: lattice dockerfile

smgl9, 2021-01-21:


umarcor, 2021-01-29:
FTR, we cannot provide pre-built images for vendor tools for two reasons: 1) licensing issues and 2) we don't want to spend time on testing and promoting non-floss tools explicitly. However, not all devices are supported by open source tooling yet; so, from a user community perspective, many of us do need to use vendor tools. Hence, we can provide a `vendors`/`non-floss` subdir, where we share dockerfiles for users to build and use in-house. That allows sharing knowledge about common issues (dependencies, starting license servers, etc.), without significantly increasing the maintenance burden.

umarcor, 2021-03-27:
Ref: https://github.com/Gekkio/docker-fpga