hdl/containers #31: No documentation around how to setup system to run docker build locally

mithro, 2021-06-12:
https://hdl.github.io/containers/#_build explains;
```
A helper script is provided in .github/bin/dockerBuild to ease building images, by hiding all common options. It’s a wrapper around command build of pyHDLC.cli:
```
But just running the script ends up with;
```
tansell@tansell-glapstation:~/github/hdl/containers$ .github/bin/dockerBuild magicls
Traceback (most recent call last):
  File "/ssd/github/hdl/containers/./pyHDLC/cli.py", line 26, in <module>
    from pyAttributes.ArgParseAttributes import (
```

Which is clearly got something to do with missing dependencies for pyHDLC but there doesn't seem to be any instructions on how to set up things to run pyHDLC.....

mithro, 2021-06-12:
Seems like you can do a;
```
virtualenv venv
source venv/bin/activate
pip install -r pyHDLC/requirements.txt
```

umarcor, 2021-06-13:
There is https://github.com/hdl/containers/blob/main/.github/setup.sh, which is specific for CI. However, you are correct in using a virtual environment for installing things through pip.

I'm unsure about publishing it through PyPI and/or just explicitly documenting how to use it locally. It feels rather specific to this repo, yet the pyHDLC alone can be used for building/maintaining container collections for other areas/domains. What do you think?  Shall we keep it as a subdir here or shall I create umarcor/pyContainerCollections?

mithro, 2021-06-13:
Basically you need instructions for
> I just did a `git clone https://github.com/hdl/containers` and now I want to build the output.

It should include step-by-step instructions for installing the required dependencies and then building a given container.

umarcor, 2021-06-13:
There is now some documentation about how to install the dependencies and set the PATH: https://hdl.github.io/containers/#_utils.

umarcor, 2022-05-20:
There is now a section in the documentation explaining the [Utils](https://hdl.github.io/containers/dev/Utils.html). pyHDLC commands/functions are automatically documented, and the syntax of the config YAML file is explained.

closed 2022-05-20 by umarcor