hdl/containers #46: Add a container for OpenROAD

tspyrou, 2021-09-21:
Please contact me if run into any issues. The repo already has a DOCKERFILE.

tspyrou, 2021-09-21:
https://github.com/The-OpenROAD-Project/OpenROAD

umarcor, 2022-02-18:
Ref: https://github.com/efabless/openlane#openlane-design-stages

umarcor, 2022-05-23:
`openroad` was added in https://github.com/hdl/containers/commit/572ef81d74118403bc0b41b60032d7701df09866 and `openroad/gui` was added in https://github.com/hdl/containers/commit/cdb9638a03c258f72d45e84a4e72735918af592c (see https://hdl.github.io/containers/dev/Graphs.html#img-graph-asic). The following screenshot was taken on Windows 10, running `openroad/gui` through x11docker and runx on MSYS2, using Docker on HyperV and VcxSrv:

[image]

```
# x11docker --runx --no-auth -i gcr.io/hdl-containers/openroad/gui                            
x11docker note: Enabled X over TCP instead of sharing unix socket.

x11docker WARNING: Option --no-auth: SECURITY RISK!
  Allowing access to new X server for everyone.
  Your X server is accessable over TCP network without any restriction.
  That can be abused to take control over your system.

Image gcr.io/hdl-containers/openroad/gui not found locally.
Do you want to pull it from docker hub?
(timeout after 60s assuming no) [Y|n]yx11docker note: Pulling image 'gcr.io/hdl-containers/openroad/gui' from docker hub

Using default tag: latest
latest: Pulling from hdl-containers/openroad/gui
214ca5fb9032: Already exists
88942047b930: Pull complete
1a074bea663f: Pull complete
d5517e695007: Pull complete
7cdd009fd2e0: Pull complete
Digest: sha256:d6041e2ef81ac09d6b6f98b17bc8354cd81ef63c0f73d155655eb245185fa16f
Status: Downloaded newer image for gcr.io/hdl-containers/openroad/gui:latest
gcr.io/hdl-containers/openroad/gui:latest
eine@e7222b42dcb5:~$ openroad -gui
```

@tspyrou, I'd be glad if you could have a quick look at https://github.com/hdl/containers/blob/main/debian-bullseye/openroad/HDLC and please let me know if you find anything wrong.

/cc @mithro @maliberty