hdl/containers #56: Xyce cannot be fetched from sandia.gov

allexoll, 2022-03-02:
Xyce image fails because it seems that https://xyce.sandia.gov/downloads/_assets/documents/Xyce-6.12.tar.gz is behind a registration page? 

I'm not sure what the fix would be for this, I'm opening this so we could track progress in case someone wants to take it on.

umarcor, 2022-03-02:
I guess we can pick tarballs from https://github.com/Xyce/Xyce/releases. However, note that we are currently using Xyce version 6 and the latest is 7. We should update. I did some attempts, but I couldn't get it finished. It should not be complex, but testing takes time. See:

- https://github.com/hdl/containers/commits/xyce
- https://github.com/google/skywater-pdk/issues/186#issuecomment-818246809

tvrusso, 2022-03-03:
The Xyce web site was migrated to a new CMS by corporate, and all deep links have been changed.  You will have to go back to square one and work out the new deep link to use.  One should consider deep links into the Xyce web site to be unreliable --- and it was in fact our intent (as mandated by upper management) that download of source and binaries from that site should never work without registration.  The old site let that sneak through if you knew the deep link, the new one relies on cookies to manage access to source and binaries.

You are better off using Github release tags to download tarballs than using deep links into the Xyce web site.   See https://github.com/Xyce/Xyce/releases.  All releases starting with 6.11 through the current release are there.

allexoll, 2022-03-04:
I see. 

I've somewhat fixed part of the build for xyce 6.12 using the tarball, but i'm hitting an error further in, and its kind of out of my scope, so i'm leaving this here if anyone else wants to take it on:

```
# Get Trilinos and Xyce
RUN mkdir -p Trilinos/trilinos-source \
 && curl -fsSL https://github.com/trilinos/Trilinos/archive/trilinos-release-12-12-1.tar.gz | \
    tar xz -C Trilinos/trilinos-source --strip-components=1 \
 && mkdir -p Xyce \
 && curl -fsSL https://github.com/Xyce/Xyce/archive/refs/tags/Release-6.12.0.tar.gz | \
    tar xz -C Xyce --strip-components=1 && \
    cd Xyce && \
    ./bootstrap   # Need to be bootstraped to generate the ./configure script for later 
```

and we also need to add the `automake` pkg to `apt-get` because `bootstrap` uses `aclocal`.

Once that is done, it can start compiling, but it fails at the end with 
```
#9 765.7   CXX      Xyce.o
#9 765.7   CCLD     libxyce.la
#9 768.7   CXXLD    Xyce
#9 769.2 /usr/bin/ld: Xyce.o: undefined reference to symbol '_ZN7Teuchos19ActiveRCPNodesSetupD1Ev'
#9 769.2 /usr/bin/ld: /tmp/xyce/usr/local/lib/libteuchoscore.so.12: error adding symbols: DSO missing from command line
```

Demangled: that symbol is `Teuchos::ActiveRCPNodesSetup::~ActiveRCPNodesSetup()`

umarcor, 2022-04-27:
It took longer than desired, but I could get to update the Xyce image in the end!

@allexoll, as you found out, changing the tarball and using autoconf/automake was not enough. I could reproduce the `undefined reference to symbol` error.

It seems not to be worth dealing with 6.12 or 7.2, so I went with 7.3 straightaway. After applying the updated guidelines [xyce.sandia.gov/documentation-tutorials/building-guide/](https://xyce.sandia.gov/documentation-tutorials/building-guide/), it worked! Then, I updated to 7.4 and last to `master`. As a result, `gcr.io/hdl-containers/xyce`, `hdlc/xyce` and `ghcr.io/hdl/xyce:latest` are now following the latest upstream.

@tvrusso thanks a lot for the detailed explanations in https://github.com/google/skywater-pdk/issues/186. I think all the issues are addressed: bison, bootstrap, update of Trilinos and Xyce build options, use ARCHDIR as a CLI argument instead of an envvar, enabe stokhos and amesos2, remove UMFPACK... I'd be glad if you could have a quick look at https://github.com/hdl/containers/blob/main/debian-bullseye/xyce.dockerfile and let me know if you find anything incorrect.

We might improve the dockerfile in the future by using cmake (https://github.com/Xyce/Xyce/blob/master/INSTALL.md), however, I believe this issue can be closed since fetching the sources is solved.

closed 2022-04-27 by umarcor