hdl/containers #70: Yosys build has failed for a long time

Topi-ab, 2024-07-04:
I was wondering why hdlc/formal from docker hub had outdated yosys build.

Now I realized that the last time build succeeded was Dec 15, 2023.
https://github.com/hdl/containers/actions/workflows/yosys.yml
https://github.com/hdl/containers/actions/runs/7216298306

Looks like g++ is missing from the preparation
https://github.com/hdl/containers/actions/runs/9705516523/job/26787730757#step:3:1431

Topi-ab, 2024-07-04:
It seems I found the issue. Tried to push a commit, but failed due missing permissions.

debian-bullseye/yosys/HDLC:

1. Add build-essential to makedepends

```makedepends=(
  bison
  flex
  gawk
  gcc
  build-essential
  pkg-config
  zlib1g-dev
  clang
  git
  make
)
```

2. Change git clone line to include submodules:

`git clone --recurse-submodules https://github.com/YosysHQ/yosys.git /tmp/yosys`

parasyte, 2024-07-17:
Today I hit two issues with the old Yosys in this container. One is that `abc` crashes with an assertion failure in some cases, and it was supposed fixed by https://github.com/YosysHQ/yosys/pull/4351 in Yosys 0.41.

The other problem I just kind of hope was resolved since December, but the `abc` one is my current blocker.